ryan-ma/PD_Micro #2: Get list of possible voltages/current limits

kaythe, 2021-06-24:
It could be very useful to get all the possible voltages and current limits the power supply supports. The debug output contains such a list, but a method to get this list would be far more useful.

danricho, 2021-07-04:
This would be great as I don't know how we can toggle between available options without an external controller.

kaythe, 2021-07-05:
I played around with it and added a method to get the information based on the way it already uses for the debug log. I'll make a pull request once I polished it up a bit.

danricho, 2021-07-05:
Awesome! Looking forward to seeing it!

danricho, 2021-07-18:
@kaythe, do you think you could share a draft of your improvement? I haven't been able to do the same :(

kaythe, 2021-07-19:
> @kaythe, do you think you could share a draft of your improvement? I haven't been able to do the same :(

I haven't had the time to work on it unfortunately, but I'll see if I can get you a rough version today

kaythe, 2021-07-20:
@danricho it took a bit longer than expected but you should now be able to see it here: https://github.com/kaythe/PD_Micro/commit/a686b230e8572daadeb7dc137b0377f7306190fc

I want to add some error handling before I create a pull request though.

danricho, 2021-07-20:
Looks great so far. And even a display output added!

I'm planning to do similar to create a "bench power supply" using either the voltage levels available or PPS if the source can do it.

I will have a tinker when I have some time at my bench.

Cheers!

gyboy, 2021-07-21:
hey,guys,it looks like we all think same things,I have made one  OLED current measure kit using INA219 and STM32 reffernced "USB Tester OLED Backpack" on github,recently I want to add the PD function so that this kit can generate any voltage and limit current while draw the scope of voltage, the PCB is done and I am debugging the software,but I have met problem with transmit the code to C while the oringinal code is C ,the software build passed while the target voltage don't change,I an still trying to find the problems.

ryan-ma, 2021-07-27:
Hi, guys, sorry for this late reply. I did write some code on getting the list of possible voltage and current. But I haven't had time to try and verify so far. 

PD micro code is designed to be small to fit into low cost embedded platform. It should be easy to just edit the PD_UFP.cpp to fit your own needs. If you need to port to other platforms, just follow PD_UFP.cpp to implement these handlers.
- PD_Protocol events handler
- FUSB302 events handler
- timer handler
- interrupt handler

I will work on porting it to Atmel SAMD51 with a new dev board with LCD this year

kaythe, 2021-07-28:
> PD micro code is designed to be small to fit into low cost embedded platform.

Yeah that's why I made the method as short as possible. It's a bit annoying to use this way but it should take up a minimum amount of time and memory. I already had to split the display initialization up into several small parts so I could use it without blocking the PD loop ^^"

If you could integrate a nicer method to access the possible voltages and currents that'd be great, otherwise my solution does everything *I*needed.

danricho, 2021-07-28:
I've had success with your method also @kaythe.

My last step is to add listing PPS mode as a selectable mode aside from the set voltages available. I'm still yet to look how I will do that, how to test for it and to get the ranges available.

Ueberspannung, 2023-08-18:
There is a simple way of getting the available PDOs. Just initialize to 5V Fix provile. When done all PDOs ar stored in the protocol structure 

I'm ausing a wrappe class derived from PD_UFP_c

`
bool 	PD_PPS_c::get_PDO_info(uint8_t PDO, PD_power_info_t * power_info)	// gets advertised voltage in mV
{ 	//	get_PDO_info 														   current in mA, power in 0,01W
	memset(power_info,0,sizeof(PD_power_info_t));
	if (PD_protocol_get_power_info(&protocol,PDO,power_info))
	{	// recalculate units
		power_info->min_v*=50;	//  50 mV units => [min_v]= 1 mV
		power_info->max_v*=50;	//  50 mV units => [max_v]= 1 mV
		power_info->max_i*=10;	//  10 mA units => [max_i]= 1 mA
		power_info->max_p*=25;	// 250 mW units => [max_p]=10 mW
		return true;
	}	// recalculate units
	else
		return false;
}	//	get_PDO_info																			
`