ryan-ma/PD_Micro #3: avrdude: butterfly_recv(): programmer is not responding

jenesaisdiq, 2021-08-10:
Hey, I'm just getting around to using one of the PD Micros, and I'm getting this error when I try to flash it. Using Arduino IDE 1.8.13 on MacOS 11.4.

I'm also seeing the onboard LEDs blinking- 5V and <3A are both blinking. Does this indicate that the bootloader needs to be reflashed?

ryan-ma, 2021-08-13:
Hi, I just tried with my M1 MacBook Air, MacOS 11.51, and yes, it fails. 
It seems with the board connected to MacBook, It cannot enter bootloader mode by all means.
Typically, double click the reset button will enter bootloader mode, and <=1A LED will start breathing. Can try to connect to the type-C power adapter and do the double click to verify.

Blinking 5V <3A means the power source only supports up to 5V <3A, which does not meet the target voltage/current.

I remember I did success program on the PD micro with MacBook. I will investigate what is going on with the new MacOS.

jenesaisdiq, 2021-08-13:
Thank you! I switched over to a Win10 desktop and it's working fine, so not a problem with this PD Micro