ryan-ma/PD_Micro #5: Add a diode between VBus and the USB-C connector?

EternityForest, 2021-08-22:
Hey! I love this board, but one thing that would really enable a lot of use cases is an additional diode that prevented backflow from VBus to the USB port, and a second diode that allowed current flow from the pin header back to the voltage regulator.

With that, we would have a choice of a few different operating modes with a battery connected to the IO header:

* Q1 turned off,  PD requesting a voltage above the battery voltage, running directly from USB

* Q1 on, PD requesting constant current, battery is charging 

* No USB power, battery is powering the voltage regulator

As far as I can tell, battery charging is one of the most exciting uses for this, and it would be amazing to use the onboard voltage regulator when on battery power.

ryan-ma, 2021-08-24:
Thanks for your suggestion. I noticed this problem after the board was completed and mass produced.
I am working on the next generation PD trigger. Stay tune in CrowdSupply.

EternityForest, 2021-09-10:
Great to hear! Sounds like this device will truly be a world first in the next rev!

doug-gilbert, 2021-11-12:
Many designs have back to back PFETs, joined at their Sources with a common Gate drive as a Vbus switch (rather than a single PFET). Would this stop the backflow you are concerned about?