ryan-ma/PD_Micro #6: Voltage present after load switch is set to off

Benjin454, 2021-08-24:
Is the load switch suppose to cut all power from vout? When PD_UFP.set_output is set to zero I'm still getting 5v. Is there a way to completely cut off voltage to vout?

ryan-ma, 2021-08-24:
Yes, the load switch is supposed to cut all power from VBUS to VOUT. Can you check if the Q1 MOSFET gate was not pull down and  Q2 Transistor Base is ON?

Benjin454, 2021-08-24:
Thanks you for the response. I actually have two pd micros. Just tried the other one and it works as expected. The soldering iron potentially was too hot on the first one when I was soldering on the screw terminal. In turn possibly damaging the load P-Channel MOSFET? Ill look into it more later. Thanks again.

closed 2022-01-04 by Benjin454

faultylee, 2022-01-04:
> The soldering iron potentially was too hot on the first one when I was soldering on the screw terminal. 
@Benjin454 It's likely not the temperature. Most soldering iron leaks, and since most MOSFET's gate behave like a capacitor and the leaking current might built up voltages higher than expected for the Gate causing it to breakdown, which then keeping it "turn-on" permanently. The solution to this is to ground the soldering iron tip, see https://electronics.stackexchange.com/questions/11583/grounding-a-soldering-iron. 

Another possibility is ESD (electro-static discharge). If you live in a dry climate or work inside a cold and dry air-conditioned room with carpet floor, this could easily happen as well, when our finger carries the high voltage static and discharged through the MOSTFET's Gate.