ryan-ma/PD_Micro #10: PPS defaults to 5V after working correctly for 60 seconds

awc105, 2022-03-18:
Hi, I have been making use of this library for a project I am working on. I have successfully run it on: nRF52832, nRF52840, and SAMD51, all cortex-M4. Using adafruit feather bootloaders for all boards. I am using a small FUSB302 breakout board from Tindie.
The PD2.0 functionality of this library works great, using 4 different adapters I was able to enable all PDOs on each adapter! 

The adapters I am using are: 
Apple 87W Charger, PD2.0 only
Wotobeus 130W Charger, PD3.0 capable of 3.3-21V @ 5A using PPS
Samsung 25W Charger, PD3.0 capable of 3.3-21V @ 1A using PPS

The cables I am using are:
Samsung 100W Marked Cable
Cable Matters 100W Marked Cable (USB-IF certified)

What chargers/cables did you use when testing?

I am also able to use PPS to set voltage and current, this works well and I have even load tested the current limiting, however on both PPS capable adapters I have, the **PPS voltage will drop out after 60-80 seconds on average and default to PD2.0**. Unplugging the cable from the charger and plugging it back in will fix the problem for another 60-80 seconds.

Do you have any suggestions or ideas as to why this is happening? 
To me it seems to be perhaps a timing issue?
Are there any parameters that I should be specifically looking at?

I have PD_UFP.run() in the main loop (using arduino) with a 1ms delay to service the state machine, I also have another version of the code where PD_UFP.run() is in a timer callback that runs at 1ms as well. Both give me the same issue.

One last small issue is that setting the PPS voltage to 21V will still yield an output of 20V on both PPS chargers even though they are capable of 21V

Thank you very much for the help!

-Andrew

awc105, 2022-03-31:
Perhaps a clue. I spent some time playing around with the timings for pps request and was able to get it to run stable for an hour, but after that it would still fault and resort to PD2.0. I then tried it using an atmega328p and it ran fine without a problem. What could be the difference?

jarrod89, 2024-10-15:
Your issue is due to the way the ARM32 compiler handles unsigned arithmetic and overflows vs the AVR compiler, I fixed this in a PR for a forked repo here, the same changes should fix this repo https://github.com/kcl93/fusb302_arduino/pull/4